MOLAorg/mola #11: build failed

Osborn28, 2019-07-17:
Hi, When I run 'cmake -H. -Bbuild', and get much error log like
'CMake Error at modules/mola-common/mola_cmake_functions.cmake:98 (target_compile_features):
  target_compile_features specified unknown feature "cxx_std_17" for target  "dbow2".
Call Stack (most recent call first):
  modules/mola-common/mola_cmake_functions.cmake:112 (mola_set_target_cxx17)
  modules/mola-common/mola_cmake_functions.cmake:253 (mola_set_target_build_options)
  externals/DBoW2/CMakeLists.txt:38 (mola_add_library)'
And I'm sure I have installed all dependency, can you help me? Thank you.

jlblancoc, 2019-07-17:
Hi @Osborn28 , 

What's your OS and OS version? What's your compiler? CMake version?

You can get all these running: 

```
lsb_release -a
c++ --version
cmake --version
```

jlblancoc, 2019-07-17:
PS: It seems you use a version of CMake older than those we test manually and on CircleCI, so it will be good to know what's your system to see if a fix can be published.

Osborn28, 2019-07-18:
Hi @jlblancoc ,
Thank you for your help. And I have solved my issue. In order to help others who have the same issue, I will summarize the issues I have encountered.
1. Install gcc7.
[https://gist.github.com/jlblancoc/99521194aba975286c80f93e47966dc5](url) (Thank you again @jlblancoc )
2. much error log like
'CMake Error at modules/mola-common/mola_cmake_functions.cmake:98 (target_compile_features):
target_compile_features specified unknown feature "cxx_std_17" for target "dbow2".
The version of CMake is old and should update. Run 'cmake --version' to check it. My version is 3.5.1, and update to 3.9.1 to slove this issue. [https://cmake.org/files/v3.9/cmake-3.9.1-Linux-x86_64.tar.gz](url)
3. /usr/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen.
GTSAM depends on Eigen3.3.*, and you can check your Eigen version by to see "/usr/include/eigen3/Eigen/src/Core/util/Macros.h" or "/usr/local/include/eigen3/Eigen/src/Core/util/Macros.h". You can download Eigen 3.3.* from [http://eigen.tuxfamily.org/index.php?title=Main_Page](url).

jlblancoc, 2019-08-09:
Closing, since the problem was fixed. Hopefully Google will still redirect people here when searching for those exact errors ;-)

closed 2019-08-09 by jlblancoc