MOLAorg/mola #14: Cmake Warning

carleft, 2019-09-19:
Hello!

I have a warning when I run `cmake ..`

`
Cannot generate a safe runtime search path for target mola-input-ros1
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libcv_bridge.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/kinetic/lib
    runtime library [libtf2.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/kinetic/lib
`
How can I solve it?
Thanks in advance!

jlblancoc, 2019-09-19:
It seems you have the ROS Kinetic setup script being called automatically in your `~./bashrc`, then you also have some ROS packages from the official Ubuntu repo (probably because you followed the MOLA build instructions?). 

I see two ways to solve this: 

A) If you plan to keep using ROS Kinetic often from the same console / apps than MOLA, uninstall ROS packages from the official Ubuntu repos: `sudo apt uninstall libroscpp-dev libsensor-msgs-dev libcv-bridge-dev`, and any other package for which you have a warning. 

B) You can remove the automatic inclusion of ROS Kinetic from bashrc (the `source /opt/ros/...`  line), and then MOLA will build against the Ubuntu official packages. From a different terminal, you can always source the ROS Kinetic file and keep using ROS. 

Personally I use (B) since that allows me to use different versions of ROS (ROS 1, ROS 2,...), and define a bash alias to quickly source the corresponding setup.bash file, but it's up to each one which way to go...

carleft, 2019-09-19:
> It seems you have the ROS Kinetic setup script being called automatically in your `~./bashrc`, then you also have some ROS packages from the official Ubuntu repo (probably because you followed the [...]

So it means that I should delete the code `source /opt/ros/kinetic/setup.bash` from ~/.bashrc ?

And if I want to use ros in /opt/ros, do I need run `source /opt/ros/kinetic/setup.bash` in the terminal?

closed 2019-09-19 by carleft

jlblancoc, 2019-09-19:
Yes, exactly, that's the option B above...

carleft, 2019-09-19:
> Yes, exactly, that's the option B above...

Thanks a lot !  It's very useful for me !