MOLAorg/mola #15: make error

carleft, 2019-09-19:
Hello!
I have a error when I making this program :
`
/home/tb/workspace/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:205:20: error: ‘ISAM2UpdateParams’ is not a member of ‘gtsam’
             ->gtsam::ISAM2UpdateParams updateParams;
`

`home/tb/workspace/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:205:20: note: suggested alternative: ‘ISAM2Params’
             ->gtsam::ISAM2UpdateParams updateParams;
`

`/home/tb/workspace/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:206:13: error: ‘updateParams’ was not declared in this scope
             ->updateParams.newAffectedKeys =
`

And I also met lots of note like this

`/home/tb/workspace/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:542:75: note: #pragma message: TODO: Disabled temporarily; see comment for `gtsam::IMUHelper `imu`

                                                                                                                    

How can I solve it?
Thank you very much in advance!

jlblancoc, 2019-09-19:
What version of GTSAM are you using? 

On disabling optimize native, it's just checking OFF the CMake flag `GTSAM_BUILD_WITH_MARCH_NATIVE` while building gtsam. I'll probably improve the docs in that area...

carleft, 2019-09-20:
> What version of GTSAM are you using?
> 
> On disabling optimize native, it's just checking OFF the CMake flag `GTSAM_BUILD_WITH_MARCH_NATIVE` while building gtsam. I'll probably improve the docs in that area...

Thank you for your reply ! 
I am using GTSAM 4.0.0 
So I need to add `set(GTSAM_BUILD_WITH_MARCH_NATIVE      OFF)` in the gtsam CMakeList.txt ?

jlblancoc, 2019-09-20:
> So I added set(GTSAM_BUILD_WITH_MARCH_NATIVE OFF) in the gtsam CMakeList.txt and rebuild gtsam. And I rebuild MOLA.

Nope... use `ccmake` or `cmake-gui` instead.

On the "ISAM2Params" error, I think that we introduced that structure into gtsam after the release of 4.0.0... Please, take the latest release, from yesterday, I think ;-)

jlblancoc, 2019-10-01:
Any news on this problem?

aaravrav142, 2019-10-03:
Hi,

I am facing a similar problem. Can you describe which fork of gtsam to use
for successful use with MOLA and how to actually build it with cmake-gui or
command prompt on ubuntu 16.

Thanks

Alex

jlblancoc, 2019-10-03:
> Can you describe which fork of gtsam to use
for successful use with MOLA and how to actually build it with cmake-gui or
command prompt on ubuntu 16.

You can use the current gtsam master branch, and build it with: 

```
cd gtsam
mkdir build
cd build
cmake .. -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF 
make gtsam && make gtsam_unstable
```

You'll need to upgrade cmake to 3.9 or newer. I prefer downloading cmake sources and building them setting `CMAKE_INSTALL_PREFIX` to some directory under home, e.g.: 

```
cd cmake-3.15.2
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/local_install_cmake/
make install -j4
$HOME/local_install_cmake/bin/cmake --version
```

Then go on with the normal compilation of `mola`:

```
cd mola
# Update the git repo if needed
cd build
# Change the build directories below as needed to match your local directories:
$HOME/local_install_cmake/bin/cmake .. -DGTSAM_DIR=$HOME/gtsam/build
```

aaravrav142, 2019-10-07:
Hi @jlblancoc 

Thank you for the detailed instructions. 
I am getting the following error when building mola now.

```
/home/arav/extra_lib/gtsam/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(

```
What version of eigen is required to successfully build with GTSAM?

Thanks

Alex

carleft, 2019-10-07:
> Any news on this problem?

I changed my ubuntu system version from 16.04 to 18.04, and there were no errors. Thanks a lot for your answer!

closed 2019-10-07 by carleft

jlblancoc, 2019-10-07:
@carleft :
> I changed my ubuntu system version from 16.04 to 18.04, and there were no errors. Thanks a lot for your answer!

Great! 

@aaravrav142 :

> Should I make gtsam with
> -DGTSAM_USE_SYSTEM_EIGEN=ON
> What version of eigen is required to successfully build with GTSAM? I want to use it with ROS kinetic on ubuntu 16

If you want to use ros, gtsam, mola, all together, probably the easiest solution is to: 
1) Make sure you have `source`d  ROS setup.bash, which most people normally do automatically via its `.bashrc` file. 
2) Then configure GTSAM with `-DGTSAM_USE_SYSTEM_EIGEN=ON`. Take a look at CMakeCache.txt to check if the version of Eigen that CMake found is under /opt/ros/..., which I guess will be the case. 
3) Build gtsam, then build mola. 
4) Build your own apps, as needed. 

All programs should now have found ROS' version of Eigen.