MOLAorg/mola #17: double free or corruption

carleft, 2019-10-10:
Hi~

Today I met a problem, When I run 
` MOLA_INPUT_RAWLOG=map1_test1.rawlog  mola-cli -c rawlog_odom_and_lidar.yml -p`, it returned

**_[21:38:02.5804|INFO |MolaLauncherApp] Setting up system from YAML config... (set DEBUG verbosity level to see full config)
[21:38:02.8805|INFO |MolaLauncherApp] Instantiating module `backend` of type `mola::ASLAM_gtsam`
[21:38:02.8811|INFO |MolaLauncherApp] Instantiating module `map` of type `mola::WorldModel`
[21:38:02.8813|INFO |WorldModel] === Using map name: `mola_map_2019_10_10,21_38_02.881325` ===
[21:38:02.8813|INFO |WorldModel] === Setting map storage base directory: `/tmp/mola_map_2019_10_10,21_38_02.881325/` ===
[21:38:02.8817|INFO |MolaLauncherApp] Instantiating module `lidar_fe` of type `mola::LidarOdometry`
[21:38:02.8821|INFO |MolaLauncherApp] Instantiating module `rawlog_input` of type `mola::RawlogDataset`
[21:38:02.8826|INFO |MolaLauncherApp] Entering main SLAM/localization loop...(CTRL+C from mola-cli to stop)
[21:38:02.8826|INFO |mola::ASLAM_gtsam:backend] Attached to WorldModel module `mola::WorldModel:map`
[21:38:02.8835|INFO |mola::LidarOdometry:lidar_fe] Attached to SLAM backend module `mola::ASLAM_gtsam:backend`
[21:38:02.8835|INFO |mola::LidarOdometry:lidar_fe] Attached to WorldModel module `mola::WorldModel:map`
[21:38:02.8836|INFO |mola::LidarOdometry:lidar_fe] Number of ICP working threads: 2 (determined automatically)
[21:38:02.9248|WARN |mola::ASLAM_gtsam:backend] Creating KF #1 without a good initial guess.
[21:38:02.9282|INFO |mola::LidarOdometry:lidar_fe] New KF: ID=1
double free or corruption (out)
Aborted (core dumped)_** in the terminal.

then the window which is mapping closed.

How can I solve it? 
 
Thanks a lot in advance!

jlblancoc, 2019-10-10:
Hi, 

Could you please find the call stack at the crash point?

Use gdb:
`MOLA_INPUT_RAWLOG=map1_test1.rawlog gdb --args mola-cli -c rawlog_odom_and_lidar.yml -p`

when it crashes, write: 

`bt`

Also, please, remember me what was your setting: OS version, whether MRPT and/or GTSAM were installed from PPA or build from sources, etc.?

carleft, 2019-10-11:
> Hi,
>
> Could you please find the call stack at the crash point?
>
> Use gdb:
> `MOLA_INPUT_RAWLOG=map1_test1.rawlog gdb --args mola-cli -c rawlog_odom_and_lidar.yml -p`
>
> when it crashes, write:
>
> `bt` [...]

Thanks for your reply!

I am using Ubuntu 18.04 and I installed MRPT and GTSAM from PPA:
```
sudo add-apt-repository ppa:joseluisblancoc/mrpt
sudo apt update
sudo apt install libmrpt-dev mrpt-apps
sudo add-apt-repository ppa:joseluisblancoc/gtsam-develop
sudo apt update
sudo apt install libgtsam-dev
```
I have tried that commend, It returned 

```
tb@tb:~/Workspace/mola/demos$ MOLA_INPUT_RAWLOG=map1_test1.rawlog gdb --args mola-cli -c rawlog_odom_and_lidar.yml -p
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from mola-cli...(no debugging symbols found)...done.
(gdb) bt
No stack.

```

jlblancoc, 2019-10-11:
Ohh, sorry I forgot: once inside gdb, you must execute "run" for the
program to actually start, then when it crashes, "bt".

Anyway, probably removing gtsam with

sudo apt remove libgtsam-dev

And building it from sources might fix it. Most often these errors are due
to optimizations built from a microprocessor architecture different than
the one in the user computer....

carleft, 2019-10-12:
> Ohh, sorry I forgot: once inside gdb, you must execute "run" for the program to actually start, then when it crashes, "bt". Anyway, probably removing gtsam with sudo apt remove libgtsam-dev And [...]

Thanks a lot! It fixed when I rebuilded gtsam form source.

closed 2019-10-12 by carleft