MOLAorg/mola #28: cmake error

lokeshluckybisht, 2020-07-09:
**On running the command** - cmake -DMOLA_BUILD_MARCH_NATIVE=OFF .. 

I am getting this error ...

-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- GTSAM found version: 4.0.2
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
OpenCL disabled, not looking for it
-- Found Boost: /usr/include (found version "1.58.0") found components: thread filesystem system program_options date_time chrono atomic 
CMake Warning (dev) at /home/driverless/.local/lib/python3.6/site-packages/cmake/data/share/cmake-3.16/Modules/FindBoost.cmake:1287 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /home/driverless/.local/lib/python3.6/site-packages/cmake/data/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  externals/mp2p_icp/3rdparty/libpointmatcher/CMakeLists.txt:120 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /usr/include (found version "1.58.0") found components: thread filesystem system program_options date_time chrono chrono atomic 
-- OpenMP found, parallel computer enabled
-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
-- API Documentation (doxygen): disabled
CMake Error at externals/mp2p_icp/CMakeLists.txt:19 (find_package):
  By not providing "Findmrpt-tfest.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "mrpt-tfest", but CMake did not find one.

  Could not find a package configuration file provided by "mrpt-tfest" with
  any of the following names:

    mrpt-tfestConfig.cmake
    mrpt-tfest-config.cmake

  Add the installation prefix of "mrpt-tfest" to CMAKE_PREFIX_PATH or set
  "mrpt-tfest_DIR" to a directory containing one of the above files.  If
  "mrpt-tfest" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/driverless/mola/build/CMakeFiles/CMakeOutput.log".
See also "/home/driverless/mola/build/CMakeFiles/CMakeError.log".

Please help on how to resolve this.

jlblancoc, 2020-07-09:
Hi, 

It seems you are missing the MRPT dependency... 
Please, take a look at the installation instructions: https://docs.mola-slam.org/latest/building.html

Cheers,

closed 2020-07-09 by jlblancoc

lokeshluckybisht, 2020-07-09:
> Hi,
> 
> It seems you are missing the MRPT dependency...
> Please, take a look at the installation instructions: https://docs.mola-slam.org/latest/building.html
> 
> Cheers,

Thanks, but the MRPT dependency is installed. Please check the output.

Reading package lists... Done
Building dependency tree       
Reading state information... Done
libmrpt-dev is already the newest version (1:1.3.2-1).
mrpt-apps is already the newest version (1:1.3.2-1).
The following packages were automatically installed and are no longer required:
  daemontools evemu-tools evtest joystick libevemu3 libvtk5.10 python-pyproj
  tcl-vtk
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 120 not upgraded.