MOLAorg/mola #29: Cmake error because of mrpt

lokeshluckybisht, 2020-07-09:
On running the command - cmake -DMOLA_BUILD_MARCH_NATIVE=OFF ..

I am getting this error  ( **MRPT dependencies are already installed**)...

-- GTSAM include directory: /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- GTSAM found version: 4.0.2
-- Could NOT find LATEX (missing: LATEX_COMPILER)
OpenCL disabled, not looking for it
-- Found Boost: /usr/include (found version "1.58.0") found components: thread filesystem system program_options date_time chrono atomic
CMake Warning (dev) at /home/driverless/.local/lib/python3.6/site-packages/cmake/data/share/cmake-3.16/Modules/FindBoost.cmake:1287 (if):
Policy CMP0054 is not set: Only interpret if() arguments as variables or
keywords when unquoted. Run "cmake --help-policy CMP0054" for policy
details. Use the cmake_policy command to set the policy and suppress this
warning.

Quoted variables like "chrono" will no longer be dereferenced when the
policy is set to NEW. Since the policy is not set the OLD behavior will be
used.
Call Stack (most recent call first):
/home/driverless/.local/lib/python3.6/site-packages/cmake/data/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
externals/mp2p_icp/3rdparty/libpointmatcher/CMakeLists.txt:120 (find_package)
This warning is for project developers. Use -Wno-dev to suppress it.

-- Found Boost: /usr/include (found version "1.58.0") found components: thread filesystem system program_options date_time chrono chrono atomic
-- OpenMP found, parallel computer enabled
-- using built-in yaml-cpp, version 0.3.0
-- text-based configuration enabled
-- API Documentation (doxygen): disabled
CMake Error at externals/mp2p_icp/CMakeLists.txt:19 (find_package):
By not providing "Findmrpt-tfest.cmake" in CMAKE_MODULE_PATH this project
has asked CMake to find a package configuration file provided by
"mrpt-tfest", but CMake did not find one.

Could not find a package configuration file provided by "mrpt-tfest" with
any of the following names:

mrpt-tfestConfig.cmake
mrpt-tfest-config.cmake

Add the installation prefix of "mrpt-tfest" to CMAKE_PREFIX_PATH or set
"mrpt-tfest_DIR" to a directory containing one of the above files. If
"mrpt-tfest" provides a separate development package or SDK, be sure it has
been installed.

-- Configuring incomplete, errors occurred!
See also "/home/driverless/mola/build/CMakeFiles/CMakeOutput.log".
See also "/home/driverless/mola/build/CMakeFiles/CMakeError.log".

Please help on how to resolve this.

jlblancoc, 2020-07-10:
From [this](https://docs.mola-slam.org/latest/building.html): 

> Build dependencies
> MRPT >= 1.9.9

From your last comment: 

> libmrpt-dev is already the newest version (1:1.3.2-1).

You have a too old version :-) 
Either build from sources, or the easier way, follow [these instructions](https://docs.mola-slam.org/latest/building.html#ubuntu-18-04) to install it from the PPA, then `apt` will "see" the new version. 

Cheers.

lokeshluckybisht, 2020-07-11:
T

> From [this](https://docs.mola-slam.org/latest/building.html):
>
>
> From your last comment:
>
>
> You have a too old version :-)
> Either build from sources, or the easier way, follow [these [...]

Thank you,

After i commented last comment, i tried building mrpt from source because even with PPA , it is showing 1.3.2 only. So the issue with mrpt is solved now.

The major issue which I had after that was building mola. I will explain with the steps.

1. I followed the install instructions given [here](https://docs.mola-slam.org/latest/building.html#ubuntu-16-04) and also [this](https://github.com/MOLAorg/mola/issues/15#issuecomment-537839096) and [this](https://github.com/MOLAorg/mola/issues/15#issuecomment-538935713) which basically points out to build using a different build directory for GTSAM and using the EIGEN prefix sor GTSAM build repectively.

2. I installed every dependencies as per the instructions. However, during the last command of "**cmake --build .**" I was getting an error building , which I have share below 

[ 93%] Building CXX object modules/mola-slam-gtsam/CMakeFiles/mola-slam-gtsam.dir/src/ASLAM_gtsam.cpp.o
In file included from /home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:24:0:
/home/driverless/mola/modules/mola-slam-gtsam/include/mola-slam-gtsam/ASLAM_gtsam.h:146:31: error: ‘FactorIndex’ is not a member of ‘gtsam’
         gtsam::FastMap<gtsam::FactorIndex, gtsam::KeySet> changedSmartFactors;
                               ^~~~~~~~~~~
/home/driverless/mola/modules/mola-slam-gtsam/include/mola-slam-gtsam/ASLAM_gtsam.h:146:31: note: suggested alternative: ‘FactorIndices’
         gtsam::FastMap<gtsam::FactorIndex, gtsam::KeySet> changedSmartFactors;
                               ^~~~~~~~~~~
                               FactorIndices
/home/driverless/mola/modules/mola-slam-gtsam/include/mola-slam-gtsam/ASLAM_gtsam.h:146:57: error: template argument 1 is invalid
         gtsam::FastMap<gtsam::FactorIndex, gtsam::KeySet> changedSmartFactors;
                                                         ^
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:124:74: note: #pragma message: TODO: Disabled temporarily to avoid segfault; also, it must be refactored
     "Disabled temporarily to avoid segfault; also, it must be refactored");
                                                                          ^
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp: In member function ‘virtual void mola::ASLAM_gtsam::initialize(const string&)’:
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:165:52: note: #pragma message: TODO: Load existing map from world model?
     MRPT_TODO("Load existing map from world model?");
                                                    ^
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:175:33: note: #pragma message: TODO: make a param
         MRPT_TODO("make a param");
                                 ^
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp: In member function ‘virtual void mola::ASLAM_gtsam::spinOnce()’:
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:188:42: note: #pragma message: TODO: Refactor into 2-3 methods
     MRPT_TODO("Refactor into 2-3 methods");
                                          ^
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:202:41: error: request for member ‘empty’ in ‘((mola::ASLAM_gtsam*)this)->mola::ASLAM_gtsam::state_.mola::ASLAM_gtsam::SLAM_state::changedSmartFactors’, which is of non-class type ‘int’
             !state_.changedSmartFactors.empty())
                                         ^~~~~
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:205:20: error: ‘ISAM2UpdateParams’ is not a member of ‘gtsam’
             gtsam::ISAM2UpdateParams updateParams;
                    ^~~~~~~~~~~~~~~~~
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:205:20: note: suggested alternative: ‘ISAM2Params’
             gtsam::ISAM2UpdateParams updateParams;
                    ^~~~~~~~~~~~~~~~~
                    ISAM2Params
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:206:13: error: ‘updateParams’ was not declared in this scope
             updateParams.newAffectedKeys =
             ^~~~~~~~~~~~
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:252:40: error: request for member ‘clear’ in ‘((mola::ASLAM_gtsam*)this)->mola::ASLAM_gtsam::state_.mola::ASLAM_gtsam::SLAM_state::changedSmartFactors’, which is of non-class type ‘int’
             state_.changedSmartFactors.clear();
                                        ^~~~~
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp: In member function ‘mola::id_t mola::ASLAM_gtsam::internal_addKeyFrame_Root(const mola::BackEndBase::ProposeKF_Input&)’:
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp:542:75: note: #pragma message: TODO: Disabled temporarily; see comment for `gtsam::IMUHelper imu`
             "Disabled temporarily; see comment for `gtsam::IMUHelper imu`");
                                                                           ^
/home/driverless/mola/modules/mola-slam-gtsam/src/ASLAM_gtsam.cpp: In member function ‘mola::id_t mola::ASLAM_gtsam::find_closest_KF_in_time(const time_point&) const’:
[... cut ...]
Directory: /home/driverless/mola/build/modules/mola-slam-gtsam/tests
"SLAM_GTSAM_basic_graph" start time: 
Output:
----------------------------------------------------------
<end of output>
Test time =   0.00 sec
----------------------------------------------------------
Test Passed.
"SLAM_GTSAM_basic_graph" end time: Jul 10 12:23 CEST
"SLAM_GTSAM_basic_graph" time elapsed: 00:00:00
----------------------------------------------------------

Now I am unable to resolve this issue. Please help.

Thanks in advance.

jlblancoc, 2020-07-12:
What version of gtsam are you using?

lokeshluckybisht, 2020-07-12:
> What version of gtsam are you using?

4.0.2

lokeshluckybisht, 2020-07-15:
It appears to me that while building, I am unable to build he **test-gtsam-basic-graph** executable file. Because, I used the prefix for the GTSAM_DIR reference which might cause the issue. Any work around for this?

jlblancoc, 2021-06-25:
If not installing GTSAM from the PPA, `GTSAM_DIR` should be set to the **build** directory where GTSAM has been compiled. 
PS: It's **not** recommended to run `sudo make install` after bulilding gtsam... or anything else, by the way ;-)

closed 2021-06-25 by jlblancoc