MOLAorg/mola #34: About 3D LiDAR SLAM from KITTI dataset

dibet, 2020-10-02:
Hi, 
It is not an issue but I would like understand more about the code inside Mola, which is fantastic.
In this sample 3D LiDAR SLAM from KITTI dataset.
I would like to know if you are using landmarks in this sample for closure loop?
Best regards
Dibet

jlblancoc, 2020-10-04:
Hi, thanks. 
Loop closure is solved using raw point clouds and simultaneously detected plane patches, via the mp2p_icp library.

dibet, 2020-10-06:
then you use the planes as landmarks?

jlblancoc, 2020-10-08:
I would say "no": there is not a **MAP** with landmarks. Planes are only used at present during keyframe-wise ICP alignment, but the "map" is just a pose graph as in most pose graph-based SLAM methods.

closed 2020-10-08 by jlblancoc

dibet, 2020-10-10:
thanks by your comments