MOLAorg/mola #53: Guidance Needed for Implementing Complete SLAM Process with KITTI Dataset

papikongjohn, 2024-06-23:
Hi   @jlblancoc,

First of all, I want to express my gratitude for your incredible work on this project. It's truly a significant contribution to the field, and I appreciate all the effort you and the team have put into it.

I am currently trying to replicate the 1.0.5 version of the project on my system (Ubuntu 20.04, ROS2 Foxy). After several days of debugging, I successfully compiled the project and managed to run the demo mentioned in issue #51 which shows the replay visualization of the provided dataset.

However, I have encountered a problem. It seems that the current version of the code does not implement a complete SLAM process but only replays and displays the data. For instance, when I run:

`ros2 launch mola_demos ros-kitti-play.launch.py kitti_sequence:=00`

it only replays the data. Given your impressive results on the KITTI leaderboard [[,](https://www.cvlibs.net/datasets/kitti/eval_odometry_detail.php?&result=e2110ded55ea27e10a0142ae3a126172c69fd006)](https://www.cvlibs.net/datasets/kitti/eval_odometry_detail.php?&result=e2110ded55ea27e10a0142ae3a126172c69fd006) I believe the project should include the full SLAM process.
Could you please provide guidance on how to achieve the complete SLAM process? Ideally, I would like to follow a tutorial to play the KITTI dataset, obtain the estimated robot poses, and save them in the KITTI dataset format.

If the current version does not support this, please let me know how I might assist in implementing this feature.

Thank you very much for your help and for your continued hard work on this amazing project.

Best regards,
Evelyn

jlblancoc, 2024-06-25:
Hi, 
That's because I'm still rushing to get the proper instructions online (they will be [here](https://docs.mola-slam.org/latest/mola_lidar_odometry.html)), together with a paper and the last code piece that is still private: [mola_lidar_odometry](https://github.com/MOLAorg/mola_lidar_odometry/). 
I'm trying to get it out within very few weeks... :crossed_fingers: 

PS: your build problems were probably due to using u20.04, right? In u22.04 or later, just using "apt install" as described in [the instructions](https://docs.mola-slam.org/latest/installing.html) should be straightforward...

jlblancoc, 2024-07-27:
Tutorials are now available online: 
https://docs.mola-slam.org/latest/

Hope it helps!

closed 2024-07-27 by jlblancoc