MOLAorg/mola #54: Mola Demo

woreom, 2024-07-01:
Hi,
Thank you for the awesome repo. I'm very new to ros and Mola. however, I can get to run any of the demos and the documentation seems to be from different versions and doesn't make sense to me. Can you provide a beginner-friendly demo and tutorial for using your work?

jlblancoc, 2024-07-01:
Hi @woreom ! 
Same answer than https://github.com/MOLAorg/mola/issues/53#issuecomment-2189059469
This project is just to see a major new release and things are moving fast. The docs online are updated everyday and the missing mola_lidar_odometry package will be released asap, together with a paper explaining its details...

woreom, 2024-07-01:
> Hi @woreom ! Same answer than [#53 (comment)](https://github.com/MOLAorg/mola/issues/53#issuecomment-2189059469) This project is just to see a major new release and things are moving fast. The docs online are updated everyday and the missing mola_lidar_odometry package will be released asap, together with a paper explaining its details...

Thanks 🙏🙏

jlblancoc, 2024-07-27:
Now, step-by-step tutorials are available online: 
https://docs.mola-slam.org/latest/

closed 2024-07-27 by jlblancoc