MOLAorg/mola #67: Localization

sesem738, 2024-09-03:
Hi,

I am very interested in using this package for mapping and localization, but in the documentation, there's nothing about localization yet. Please, how do I go about this?

jlblancoc, 2024-09-04:
Hi! I know, it's one of the missing parts to document well.
For the records, the page for localization is: https://docs.mola-slam.org/latest/localization.html

I'll try to add docs and to put together a demo and/or launch file for localization ASAP, and come back here to notify about it.

PS: Did mapping go well for you already?

jlblancoc, 2024-09-07:
@sesem738 The tutorial for using a particle filter for localization, from MOLA-based maps, is now [out](https://docs.mola-slam.org/latest/localization.html#localization-with-particle-filtering).

I'll add soon the missing part of using MOLA-LO itself for localization.

sesem738, 2024-09-09:
Thank you very much. I will try the localization and let you know, but getting the mapping to work was very straightforward. I tried the ROS2 live mode. I have a question about modifying the topic for transform listeners. Currently, the transforms from my robot are being published in a specific namespace. However, the node is listening to transforms from the '/tf' topic. Is there a way to configure the node to listen to transforms from a different topic or namespace?

jlblancoc, 2024-09-09:
That's a tricky part that was not handled here yet, you are right.

Take a look at how it's solved in Nav2 (and the comments therein): 
https://github.com/ros-navigation/navigation2/blob/3ab6312ee66de07a3aa77273980f8c379224ccf0/nav2_bringup/launch/bringup_launch.py#L53-L59

We would need to add those remaps, together with the machinery for namespaces:
- https://github.com/ros-navigation/navigation2/blob/3ab6312ee66de07a3aa77273980f8c379224ccf0/nav2_bringup/launch/bringup_launch.py#L41-L42
- Push: https://github.com/ros-navigation/navigation2/blob/3ab6312ee66de07a3aa77273980f8c379224ccf0/nav2_bringup/launch/bringup_launch.py#L145

All those changes should go [here](https://github.com/MOLAorg/mola_lidar_odometry/blob/develop/ros2-launchs/ros2-lidar-odometry.launch.py), which is the launch file you used, as I guess. 

If you have a robot setup with namespaces and want to try the changes yourself and they work, a pull request would be more than welcome! ;-)

jlblancoc, 2024-12-17:
Namespaces for /tf and sensor topics are now supported. See discussion in #71. Testing and feedback by users would be welcome before making a new release!

closed 2024-12-17 by jlblancoc