MOLAorg/mola #72: ROS2 | GUI Interface Disable

SOHAILMBS, 2024-12-12:
Dear @Mola team,

Is it possible to disable the GUI interface that starts with the ros2 launcher file?

jlblancoc, 2024-12-12:
Hi, 

Mmm... if you are running from a local clone of the repo, just remove these two sets of lines: 

- https://github.com/MOLAorg/mola_lidar_odometry/blob/develop/mola-cli-launchs/lidar_odometry_ros2.yaml#L12-L15
- https://github.com/MOLAorg/mola_lidar_odometry/blob/develop/mola-cli-launchs/lidar_odometry_ros2.yaml#L40-L43

and the GUI will go out. 

Let's keep this issue open to remember adding some launch flag to disable it more easily.

SOHAILMBS, 2024-12-13:
Thanks, just what we were looking for!

jlblancoc, 2024-12-17:
For the records: 

This is now a feature in mola-lo ROS2 node, please refer to the docs (Open the tab "all launch arguments"): 
https://docs.mola-slam.org/latest/ros2api.html#ros-2-node-for-lidar-odometry-lo

New ros2 launch arguments: ``use_mola_gui:=False`` & ``use_rviz:=False``

You will need to build both MOLAOrg/mola and MOLAOrg/mola_lidar_odometry from sources after updating the source git repos, or wait (weeks...) for the official binaries from ROS build farms. 

I'll make a new release after #71  (WIP already).