MOLAorg/mola #79: No mapping on Livox MID360

Joaquin-Navas, 2025-03-21:
Hi,

I am currently using LIVOX MID360 with other Lidar Odometry systems but I am not able to make it work with MOLA.

Currently I got to make it receive real time point cloud via:

ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py \
lidar_topic_name:=/livox/lidar \
imu_topic_name:=/livox/imu \
ignore_lidar_pose_from_tf:=True \
publish_localization_following_rep105:=False

but as it can be seen below, it just fixes some points from the first cloud (the ones coloured) and shows the real time cloud (in white), but without mapping no localization. I can see ICP quality is 0% that is probably related with the issue.

Thanks!

[image]

jlblancoc, 2025-03-21:
Hey, 
Probably the same issue than https://github.com/MOLAorg/mola_lidar_odometry/issues/8

To confirm, try running it like: 

```
MOLA_SKIP_DESKEW=True \
ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py
  lidar_topic_name:=/livox/lidar
  imu_topic_name:=/livox/imu
  ignore_lidar_pose_from_tf:=True
  publish_localization_following_rep105:=False
```

and if it works, it means we should eventually address that timestamping issue for this model :-) 

In the meanwhile, you can go on with that env variable. Map quality will be a bit worse, if moving slow, or significantly worse if moving it aggressively, though...

Joaquin-Navas, 2025-03-23:
Hi,

Iḿ so sorry, I searched for some mid360 related topics but didn't see that thread.

MOLA_SKIP_DESKEW=True make it works. I will try to change the way livox ros driver 2 publishes the timestamps so that no MOLA_SKIP_DESKEW is needed.

I will change to the other thread. You can close this one.

Thanks!

jlblancoc, 2025-03-23:
No worries! It's confusing to know against what repo an issue should apply.