MOLAorg/mola #80: 2D occupancy maps not published to ROS

Saysong, 2025-04-18:
Device: Ubuntu 24.04.2 LTS
ROS: Jazzy
Mola Version: 1.6.3

When running the ROS 2 mola_lidar_odometry package with the 2D pipeline, there seems to be an issue where the ros2_bridge receives a `mrpt::maps::CVoxelMap` rather than a `mrpt::maps::COccupancyGridMap2D`. This prevents it from publishing the expected grid map to ROS. 

```
root@gsfm:~/ros2_ws# ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py     
    lidar_topic_name:=/cloud     
    ignore_lidar_pose_from_tf:=False     
    imu_topic_name:=/imu/data_raw     
    use_mola_gui:=True    
    publish_localization_following_rep105:=False     
    mola_lo_pipeline:=/opt/ros/jazzy/share/mola_lidar_odometry/pipelines/lidar2d.yaml     
    generate_simplemap:=False     
    use_state_estimator:=False     
    use_rviz:=False 
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-04-18-17-58-07-054397-gsfm-3102
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mola-cli-1]: process started with pid [3105]
[mola-cli-1] [17:58:07.6470|INFO |MolaLauncherApp] Setting up system from YAML config... (set DEBUG verbosity level to see full config)
[mola-cli-1] [17:58:07.6474|INFO |MolaLauncherApp] Instantiating module `viz` of type `mola::MolaViz`
[mola-cli-1] [17:58:07.6528|INFO |MolaLauncherApp] Instantiating module `lidar_odom` of type `mola::LidarOdometry`
[mola-cli-1] [17:58:07.6541|INFO |MolaLauncherApp] Instantiating module `state_estimation` of type `mola::state_estimation_smoother::StateEstimationSimple`
[mola-cli-1] [17:58:07.6548|INFO |MolaLauncherApp] Instantiating module `ros2_bridge` of type `mola::BridgeROS2`
[mola-cli-1] [17:58:07.6558|INFO |mola::LidarOdometry:lidar_odom] Attached to a VizInterface module
[mola-cli-1] [17:58:07.6559|INFO |MolaLauncherApp] ╔═══════════════════════════════════════╦═════════════════════════════════╗
[mola-cli-1] [17:58:07.6559|INFO |MolaLauncherApp] ║  Entering main MOLA application loop  ║ > CTRL+C for mola-cli to quit < ║
[mola-cli-1] [17:58:07.6559|INFO |MolaLauncherApp] ╚═══════════════════════════════════════╩═════════════════════════════════╝
[mola-cli-1] [17:58:07.6559|INFO |StateEstimationSimple] reset() called
[mola-cli-1] [17:58:07.6981|WARN |LidarOdometry] No YAML entry 'observations_filter_adjust_timestamps', this filter stage will have no effect.
[mola-cli-1] [17:58:07.7043|WARN |LidarOdometry] No YAML entry 'observations_filter_2nd_pass', this filter stage will have no effect.
[mola-cli-1] [17:58:07.7044|WARN |LidarOdometry] No YAML entry 'observations_filter_final_pass', this filter stage will have no effect.
[mola-cli-1] [17:58:07.7074|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA localization source module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [17:58:07.7074|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA map source module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [17:58:07.7075|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA relocalization module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [17:58:07.7481|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA map server module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [17:58:07.7584|WARN |MolaLauncherApp] Could not achieve desired real-time execution rate (20 Hz) on thread for sensor named: ros2_bridge
[mola-cli-1] [17:58:17.3459|INFO |StateEstimationSimple] reset() called
[mola-cli-1] [17:58:17.3462|INFO |LidarOdometry] Initial re-localization done with pose: (x,y,z,yaw,pitch,roll)=(0.0000,0.0000,0.0000,0.00deg,0.00deg,0.00deg)
[mola-cli-1] [17:58:17.7699|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:37.7696|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:38.7694|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:39.7696|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:40.7694|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:41.7695|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:42.7695|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:43.7697|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:44.7695|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:46.7695|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:47.7700|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:49.7700|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:50.7697|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:51.7696|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:53.7695|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:54.7703|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:55.7699|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
[mola-cli-1] [17:58:56.7694|WARN |mola::BridgeROS2:ros2_bridge] Do not know how to publish map layer 'localmap' of type 'mrpt::maps::CVoxelMap'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[mola-cli-1] [INFO] [1744999198.125033484] [rclcpp]: signal_handler(signum=2)
[mola-cli-1] Caught signal 2. Shutting down...
[mola-cli-1] tcache_thread_shutdown(): unaligned tcache chunk detected
[ERROR] [mola-cli-1]: process has died [pid 3105, exit code -6, cmd '/opt/ros/jazzy/lib/mola_launcher/mola-cli /opt/ros/jazzy/share/mola_lidar_odometry/mola-cli-launchs/lidar_odometry_ros2.yaml --ros-args -r /tf:=tf -r /tf_static:=tf_static'].
[INFO] [launch]: process[mola-cli-1] was required: shutting down launched system
```

The map is being created and the visual comes out just fine on the mola gui 

[image]

Notes:
2D LiDAR data is coming in as the /cloud topic as LiDAR driver supports sending messages in PointCloud2 format. 
Same issue arises if using /scan topic LaserScan message and edit to the `mola_lidar_odometry/mola-cli-launchs/lidar_odometry_ros2.yaml lidar_odometry_ros2.yaml` file to accept LaserScan msgs rather than PointCloud2 msgs

jlblancoc, 2025-04-19:
That's because the 2D pipeline uses a generic VoxelMap, not a 2D gridmap. 
A publisher for CVoxelMap could be implemented for ros2_bridge [here](https://github.com/MOLAorg/mola/blob/develop/mola_bridge_ros2/src/BridgeROS2.cpp#L1366-L1381), but I guess it would be not very efficient to publish all empty & free voxels (I mean: it would slow down the whole thing). An alternative is to just publish occupied voxels as a pointcloud. At present I don't have time to implement this feature, but Pull Requests adding it are welcome! :-) 

That's regarding mapping time. Now, if you don't actually care with viewing the 2D gridmap **DURING MAPPING**, and only worry about the final gridmap, you can just build the 2D (not 3D) gridmap in postprocessing, then publish it using either mrpt_map_server or MOLA-LO in localization mode. 
This is explained in [this section](https://docs.mola-slam.org/latest/building-maps.html#build-metric-maps-and-visualize-them) of the docs; see the "tab" titled "Build a voxel map + 2D grid map".

ahpinder, 2025-04-21:
Hey, I worked with Saysong and created a PR that should solve this issue. It can be found [here](https://github.com/MOLAorg/mola/pull/82).
I chose the path to export occupied voxels as points, since it seemed the most straightforward.

jlblancoc, 2025-04-21:
Good idea, straight to the point :+1:

PR #82 (2025-04-22): Add Support for Voxel Map ROS2 Publishing Via Point Map Conversion
This code adds the ability to publish point maps of occupied voxels in the ROS2 bridge, allowing for real-time visualization of 2D map generation. Since it only publishes occupied voxels, it is not overly computationally expensive.
It uses an existing MRPT function to get the points, so no custom, potentially slow point-wise code is involved.

An example RVIZ2 visualization of a map in the process of being created is shown here:
[image]

Closes #80
closed 2025-04-22 by jlblancoc